seeers/PW-Stick #1: [Request]Schematic source file

matthgyver, 2017-09-18:
Hi,

Your project seem very interresting.
I just buy an Arduino Leonardo to try it but I plan to build one from scratch.
Could you publish the source of the electronic schematic and / or the PCB layer please ?

Thank you

seeers, 2017-09-20:
Hi, 
yes of course:
[PW-StickPublic.zip](https://github.com/seeers/PW-Stick/files/1318209/PW-StickPublic.zip)

matthgyver, 2017-09-20:
Hi,
Thank you very much.
I see that there is no value for C2 and C3. What do you recommend please ?
On the way, where did you get your printed circuit boards ?

seeers, 2017-09-20:
You're welcome. C2 and C3 are 100n 
I got my Boards from https://www.itead.cc/open-pcb/pcb-prototyping.html

matthgyver, 2017-09-20:
Price look good and the finish too when I see your pictures.
What is the height of your microswitch to go with the 3D printed case please  ? (I prepare my shopping list)

matthgyver, 2017-12-29:
I've take this one : https://fr.aliexpress.com/item/Free-Shipping-100pcs-Tact-Switch-DIP-type-6x6X8-mm-Tactile-Switch/866454082.html?spm=a2g0s.9042311.0.0.RjY8lq
It's 6x6x8mm and fit nice

closed 2017-12-29 by matthgyver