seeers/PW-Stick #3: [Question] Bootloader

matthgyver, 2017-11-20:
Hi,

Does I need to burn the Arduino bootloader ?
I've tested with the default Atmega bootloader and upload scketch with Arduino IDE .
Upload was OK but the stick does'nt work.

I've read lot of things about burning arduino bootloader on Atmega 32u4 and make lot of tests but I can't burn the bootloader.
How have you proceed please ?

Thank you

seeers, 2017-11-30:
Hi, 

yes you need to burn the Arduino Leonardo Bootloader to the Atmega32u4. 
Just connect the ISP Pins to a USPASP or a similar ISP and burn the Bootloader with the Arduino IDE.

matthgyver, 2017-12-29:
I take my head with a bad programmer and some other things.
I've converted an arduino as an ISP programmer and this tutorial help me : https://murchlabs.com/monday-experiment-bootloading-an-atmega32u4-with-arduino/
All is ok now ! Work perfectly !
Thank you Seers ;-)

closed 2017-12-29 by matthgyver