seeers/PW-Stick #5: Bug on long press

matthgyver, 2018-04-22:
Hi,
I'm using PW-Stick from more 6 month and I did not understand why sometime, **on a long press**, not my password was typed. (some char are typed but not my password)
Today, working on code, I'm able to define the context : 
- Unlock PW-Stick
- Do a long press
- Result => NOK

Now : 
- Unlock PW-Stick
- Do a short press
- Login and password are typed
- Do a long press
- Password are typed
- Repeat a long press => OK

I suppose that the password if typed encrypted as long as a short press is'nt made.
I' tried to correct the bug but my knowledge is too limited.

Could you take a look please ?
Thank You

tinker4545, 2018-04-22:
matthgyver your lucky I cant even compile the sketch, would you be so kind and try and help me out, im getting errors....

Documents\Arduino\libraries\AESLib\AESLib.c: In function 'aes128_enc_multiple':
Documents\Arduino\libraries\AESLib\AESLib.c:75: error: 'for' loop initial declaration used outside C99 mode
\Documents\Arduino\libraries\AESLib\AESLib.c: In function 'aes256_enc_multiple':
\Documents\Arduino\libraries\AESLib\AESLib.c:95: error: 'for' loop initial declaration used outside C99 mode
\Documents\Arduino\libraries\AESLib\AESLib.c: In function 'aes128_dec_multiple':
Documents\Arduino\libraries\AESLib\AESLib.c:213: error: 'for' loop initial declaration used outside C99 mode
C:\Users\Cookie\Documents\Arduino\libraries\AESLib\AESLib.c: In function 'aes256_dec_multiple':
\Documents\Arduino\libraries\AESLib\AESLib.c:233: error: 'for' loop initial declaration used outside C99 mode

matthgyver, 2018-04-23:
Hi,
Use Arduino <= 1.0.5.
Visibly, there is a problem with the AES lib and the Arduino recent version.
Please, open an issue for problem other that this one.

seeers, 2018-05-03:
I think I fixed the bug. It is not tested yet

matthgyver, 2018-05-07:
Thank you
I'll test this week and give you a feedback

matthgyver, 2018-05-08:
Seers,

I've burned my Atmega directly with a TQFP32 to DIP 28 adapter and a programer and now that the stick is build, I can't burn it. Is it normal ?
So I would like to put the modification to the desktop version to do some tests but I need to update the code and I don't see where have you make your modifiactions. A diff between lastests version tell me that the files are totally different.
Could you tell me where to see please ?

Thank you

tinker4545, 2018-05-14:
Well done you too ;-) I just uploaded it, seems to be working fine :-)

closed 2019-04-17 by matthgyver