seeers/PW-Stick #9: FAO Matt

tinker4545, 2018-05-06:
Hi Matt, 

I see Seers put a fix up, just wondering are you going to update your desktop version with the fix? (if it works ;-)

also, In your desktop version I see you have keyboard.h defined, is there anyway I can get the UK keyboard to work?

Thx

matthgyver, 2018-05-07:
Hi,

I've not the time to update the desktop version. Maybe this week-end.
I think that the UK keyboard will work. The keyboard definition is on HID.cpp file (into the arduino requirement folder)

matthgyver, 2018-05-08:
Hi,

I'm french and dont well understand your problem, even with google translate :-/
I'm using a french keyboard.

tinker4545, 2018-05-08:
ok matt no worries, forget about the keyboard. I think the bug is more important for now :-)

matthgyver, 2018-09-11:
Hi,
Code was updated but I forgot to close this issue.

closed 2018-09-11 by matthgyver