seeers/PW-Stick #10: Plugins problems with linux ( keepass 2.38 )

Debian-10, 2019-02-18:
Hello and thanks again for providing this github !

I installed Keepass 2.38 with ubuntu repositories ( it uses mono ). I copied the PWstick.dll plugin into the correct path. When i restart KeePass , i have " non compatible plugin ". The thing I don't understand is that it works like a charm on windows with the same keepass version. Any ideas ?

Debian-10

Debian-10, 2019-05-09:
After reading more the code and searched more about arduino, i found solution for buttons.

closed 2019-05-09 by Debian-10

matthgyver, 2019-05-11:
Salut,

Désolé de ne pas avoir répondu, je ne reçois pas de notifications sur les issues (faut que je regarde de plus près).

Pour Mono je ne peux pas te répondre, je n'ai pas testé. J'aimerais une solution plus universelle que Keepass car çle plugin ne fonctionne que sous windows mais je n'ai pas les connaissances pour développer quelque chose :-/

Concernant le cablage je n'ai pas bien saisi ce que tu veux dire désolé.