seeers/PW-Stick #12: Problem with the plugin on windows

Debian-10, 2019-05-09:
Hello again,
As i'm still trying to get this working, i successfully burned the code into my arduino pro-micro.
I have keepass 2.41 installed on my computer and put the pwkeeper.dll plugin in the plugin/ folder.
I connect it to windows, I type the default password ' 0000 ' and unlocked the key but when trying to change settings of the key or reading its data, i always have ' No key found '.
when i press "0" button on my key, it locks the computer like it should do.
I precise that i already tried on another computer.
Thanks again for this recent update !

Debian-10

tinker4545, 2019-05-09:
The Plugin is compiled for KeePass 2.32

Debian-10, 2019-05-10:
> The Plugin is compiled for KeePass 2.32

Regarding the wiki , the plugin works on 2.41 too.

tinker4545, 2019-05-10:
Ah ur right im on the old version, try 2.32 anyways just to see

Debian-10, 2019-05-11:
What is your version of keepass when the plugin work for you ?

matthgyver, 2019-05-11:
Hi,

It's work with the last version of Keepass (a friend have tested recently).

Have you installed the Arduino IDE on your windows ? It act as a "driver" and it's needed to configure the stick with Keepass. *(only for configure it, you can use it on any computer without IDE)*

What version did you use ? (stik or desktop)

Debian-10, 2019-05-11:
Hello, i have the arduino ide on my computer.
The most strange fact is that i tested the key month ago and it worked but now not . ( Even with same arduino version and same keepass )

Debian-10, 2019-05-11:
I don't understand at all , i previously reinstalled the arduino ide but i did it now again and it works. But for how many time ?

tinker4545, 2019-05-11:
> s

2.32   anyways glad u got it working

matthgyver, 2019-05-12:
Strange. Maybe due to system update.

Debian-10, 2019-05-12:
So now , if I want to connect my key with keepass on another computer, should I install arduino on it ?

matthgyver, 2019-05-12:
Yes.
Arduino IDE contain a driver that allow serial communication between computer and Arduino.
Maybe driver installation (https://www.arduino.cc/en/Guide/DriverInstallation) is enougth. I have'nt tested.

Debian-10, 2019-05-12:
Thanks a lot for help !

closed 2019-05-12 by Debian-10